DFKI-NI/mir_robot #137: Problems with chrony on MIR500

InigoMoreno, 2024-05-08:
Hello,
I have been having some issues setting up chrony on a MIR500. When it is installed, the laser scanners stop working.
I see that the commit https://github.com/DFKI-NI/mir_robot/pull/130/commits/bf75c56e8c7b9b011613d30370c1addcf05829f5 by @pumablattlaus mentions some laser issues when adding the last step of the chrony.conf configuration. 

I share my chrony.conf in both cases:
For the `external-pc` (192.168.12.30)
```properties

confdir /etc/chrony/conf.d

pool ntp.ubuntu.com        iburst maxsources 4
pool 0.ubuntu.pool.ntp.org iburst maxsources 1
pool 1.ubuntu.pool.ntp.org iburst maxsources 1
pool 2.ubuntu.pool.ntp.org iburst maxsources 2

sourcedir /run/chrony-dhcp
sourcedir /etc/chrony/sources.d
keyfile /etc/chrony/chrony.keys
driftfile /var/lib/chrony/chrony.drift
ntsdumpdir /var/lib/chrony
logdir /var/log/chrony
maxupdateskew 100.0
rtcsync
makestep 1 3
leapsectz right/UTC

# Allow local connections
local stratum 8
allow 192.168.12.0/24

```

For the `mir-pc`:

```properties
server 192.168.12.30 iburst prefer
# Disable default pool of servesr
# pool 2.debian.pool.ntp.org offline iburst

keyfile /etc/chrony/chrony.keys
commandkey 1
driftfile /var/lib/chrony/chrony.drift
log tracking measurements statistics
logdir /var/log/chrony
maxupdateskew 100.0
dumponexit
dumpdir /var/lib/chrony
allow 192.168.12.0/24 nomodify notrap nopeer
logchange 0.5
hwclockfile /etc/adjtime
rtcsync

```

As you can see, the chrony.conf in the mir-pc already has the line `allow 192.168.12.0/24 nomodify notrap nopeer`, which is the one suggested by @pumablattlaus. Can you see anything wrong with my chrony configuration?

pumablattlaus, 2024-05-15:
After adding the line `allow 192.168.12.0/24 nomodify notrap nopeer`, the laserscanners were working for us. I will share our chrony.conf files (we have the external pc on another subnet):

**client mir:**
```bash
keyfile /etc/chrony/chrony.keys

# This directive sets the key ID used for authenticating user commands via the
# 'chronyc' program at run time.

commandkey 1

# I moved the driftfile to /var/lib/chrony to comply with the Debian
# filesystem standard.

driftfile /var/lib/chrony/chrony.drift

# Comment this line out to turn off logging.

log tracking measurements statistics
logdir /var/log/chrony

# Stop bad estimates upsetting machine clock.

maxupdateskew 100.0

# Dump measurements when daemon exits.

dumponexit

# Specify directory for dumping measurements.

dumpdir /var/lib/chrony

# This directive lets 'chronyd' to serve time even if unsynchronised to any
# NTP server.

#local stratum 10

# This directive designates subnets (or nodes) from which NTP clients are allowed
# to access to 'chronyd'.

#allow foo.example.net
#allow 10/8
#allow 0/0 (allow access by any IPv4 node)
#allow ::/0 (allow access by any IPv6 node)

# This directive forces `chronyd' to send a message to syslog if it
# makes a system clock adjustment larger than a threshold value in seconds.

logchange 0.5

# This directive defines an email address to which mail should be sent
# if chronyd applies a correction exceeding a particular threshold to the
# system clock.

# mailonchange root@localhost 0.5

# This directive tells 'chronyd' to parse the 'adjtime' file to find out if the
# real-time clock keeps local time or UTC. It overrides the 'rtconutc' directive.

hwclockfile /etc/adjtime

# This directive enables kernel synchronisation (every 11 minutes) of the
# real-time clock. Note that it can’t be used along with the 'rtcfile' directive.

rtcsync
server 10.145.8.50

# Clients from this subnet have unlimited access, but only if
# cryptographically authenticated.
#restrict 192.168.12.255 mask 255.255.255.0 nomodify notrap nopeer
allow 192.168.12.0/24 nomodify notrap nopeer

```

**external server:**
```bash
# Welcome to the chrony configuration file. See chrony.conf(5) for more
# information about usuable directives.

# This will use (up to):
# - 4 sources from ntp.ubuntu.com which some are ipv6 enabled
# - 2 sources from 2.ubuntu.pool.ntp.org which is ipv6 enabled as well
# - 1 source from [01].ubuntu.pool.ntp.org each (ipv4 only atm)
# This means by default, up to 6 dual-stack and up to 2 additional IPv4-only
# sources will be used.
# At the same time it retains some protection against one of the entries being
# down (compare to just using one of the lines). See (LP: #1754358) for the
# discussion.
#
# About using servers from the NTP Pool Project in general see (LP: #104525).
# Approved by Ubuntu Technical Board on 2011-02-08.
# See http://www.pool.ntp.org/join.html for more information.
pool ntp.ubuntu.com        iburst maxsources 4
pool 0.ubuntu.pool.ntp.org iburst maxsources 1
pool 1.ubuntu.pool.ntp.org iburst maxsources 1
pool 2.ubuntu.pool.ntp.org iburst maxsources 2

# This directive specify the location of the file containing ID/key pairs for
# NTP authentication.
keyfile /etc/chrony/chrony.keys

# This directive specify the file into which chronyd will store the rate
# information.
driftfile /var/lib/chrony/chrony.drift

# Uncomment the following line to turn logging on.
#log tracking measurements statistics

# Log files location.
logdir /var/log/chrony

# Stop bad estimates upsetting machine clock.
maxupdateskew 100.0

# This directive enables kernel synchronisation (every 11 minutes) of the
# real-time clock. Note that it can’t be used along with the 'rtcfile' directive.
rtcsync

# Step the system clock instead of slewing it if the adjustment is larger than
# one second, but only in the first three clock updates.
makestep 1 3

allow 10.145.8/24
```

InigoMoreno, 2024-05-20:
In the end I opted for using ntp instead of chrony for the clock synchronization, so as not to break the existing configuration.

closed 2024-05-20 by InigoMoreno