DFKI-NI/mir_robot #138: register new sensor

odai93sh, 2024-06-25:
Hello,
am trying to add a new radar sensor (https://github.com/uos/rmagine_gazebo_plugins)
I've followed all the instruction on the new sensor documentation, and it works with examples.
When I tried to implement it with mir_250, it dosen't works, because one step is necessary: "register a new sensor in gzserver", would you please advice me where should i register this sensor?
the command is:
xml
<node name="gazebo" pkg="gazebo_ros" type="gzserver" args="-s librmagine_embree_sensors_gzregister.so" output="screen" /> -->

mintar, 2024-06-25:
Sounds like a you should open this issue in [rmagine_gazebo_plugins](https://github.com/uos/rmagine_gazebo_plugins)), not here.

@aock: Any ideas?

mintar, 2024-06-26:
Closing this issue. Please reopen at one of @aock's repositories.

closed 2024-06-26 by mintar