DFKI-NI/mir_robot #139: MIR bridge in ROS, topic mir_msgs does not work.

MDario97, 2024-08-21:
Good morning,
i am controlling a MIR100 with ROS Noetic 20.04 installed in a virtual box (Oracle). when i launch the mir.launch file i can see all the topics created by it, and i see a strange thing: a lot of topics have as message type 'mirMsgs', all of these topics are not available for me and i can't even visualize them with the echo funcion, so i went into the  CMakeLists.txt, package.xml and mir_bridge.py files in the mir_driver folder and i noticed that mirMsgs is never present, but instead there is the message type 'mir_msgs' that  is assigned as message type of all the topics that in the terminal of the mir_bridge have 'mirMsgs' as message type. i will post some photos:
One of the Topics created after launching 'roslaunch mir_bridge mir.launch' on terminal:
[image]
Contents of mir_bridge.py file in mir_driver/nodes folder:
[image]
Contents of CMakeLists.txt file in mir_driver folder:
[image]
Contents of package.xml file in mir_driver folder:
[image]
How could i change the message type that i see in the Terminal? Thank you

mintar, 2024-08-21:
On the internal ROS system inside the MiR platform, there is a package called `mirMsgs`. In order to publish these messages to the external ROS system, I had to recreate that package and copy the message definitions. The name `mirMsgs` does not follow the naming conventions, which generates a warning in ROS1 and is even impossible in ROS2. Therefore, I've called the "external" ROS package that I created `mir_msgs`.

The difference in naming is not a problem. If there were a difference in the actual contents of the messages, you would get an error.

Echoing the `robot_mode` and `robot_state` topics works for me:

```
$ rostopic echo /robot_mode
robotMode: 7
robotModeString: "Mission"
---

$ rostopic echo /robot_state
robotState: 4
robotStateString: "Pause"
---
```

This is on MiR software version 2.8.3.1. It is possible that they've changed these topics again in a later software version.

MDario97, 2024-08-21:
If i try to echo those commands i get the error:
ERROR: Cannot load message class for [mir_msgs/Robot...]. Are your messages built?

mintar, 2024-08-21:
That means you didn't install the `mir_msgs` package, or installed from source but didn't compile it / didn't source the proper `setup.bash`. How did you install the mir_robot package? The easiest way is `sudo apt install ros-noetic-mir-robot`. That should automatically install `mir_msgs` as well.

MDario97, 2024-08-21:
These are my exact passages to install the MIR package:

1) sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
wget http://packages.ros.org/ros.key -O - | sudo apt-key add -
2)sudo apt-get update -qq
It will give an error; it will show a pub key
3)sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys <PUBKEY shown in the error>
4)sudo apt-get update
5)mkdir -p ~/catkinMIR_ws/src
6)cd ~/catkinMIR_ws/src/
7)git clone -b noetic https://github.com/DFKI-NI/mir_robot.git
8)sudo apt-get update -qq
9)sudo apt-get install -qq -y python3-rosdep
10)sudo rosdep init
11)rosdep update
12)rosdep install --from-paths ./ -i -y --rosdistro noetic
13)source /opt/ros/noetic/setup.bash
14)catkin_init_workspace
15)cd ~/catkinMIR_ws
16)catkin_make -DCMAKE_BUILD_TYPE=RelWithDebugInfo
17)source ~/catkinMIR_ws/devel/setup.bash
18)echo "source ~/catkinMIR_ws/devel/setup.bash" >> ~/.bashrc
19)source ~/.bashrc

mintar, 2024-08-21:
Glad that it works! Your installation steps from source also look correct, that should also have worked...

MDario97, 2024-08-22:
Good morning,
In these days i am moving the MIR100 with matlab, publishing and subscribing topics, and i noticed that this particular message type ('mirMsgs/...) gives problems also in Matlab/Simulink (even with the new installation, if i see those topics in ROS i don't have problems anymore), in fact when i connect to the MIR/ROS master (i have the same problem), and the i digit 'rosmsg list' in matlab, i don't have the messages mirMssgs or mir_msgs in the available ones.
In the next screens i am connected to the MIR master using 'rosinit(http://....:11311):
[image]
As we can see the topic is present and the message type is the correct one, but then if i try to read the message types available:
[image]
I don't get how it is possible, and the problem is the same even if i connect to the ROS master instead of the MIR master.

mintar, 2024-08-22:
Yes, if you connect to the MiR ROS master directly, then you will need the package `mirMsgs` from the MiR platform. I would recommend using this repository here with the mir_bridge instead.