DFKI-NI/mir_robot #140: Unable to Reset Robot Odometry

huyhoang251296, 2025-01-21:
I am currently working on applying Reinforcement Learning (RL) to the MIR robot in the EmptyWorld. As part of my RL setup, I need to reset both the world and the robot's odometry at the start of each episode to ensure consistent starting conditions.

However, I have not been able to find a way to reset the robot's odometry effectively during the simulation.

## Expected Behavior:

The robot's odometry should be reset to a predefined state (e.g., zeroed or to a specific position) at the beginning of each new episode.

## Actual Behavior:

- Even though the robot's position in Gazebo is reset, the odometry values (via /odom topic) are not reset, causing the robot to maintain its previous odometry data across episodes.
- The robot continues with its previous odometry values even when its position is set back to the original location in Gazebo, leading to inconsistencies in the simulation.

Any guidance on how to reset the robot's odometry, or an explanation of the correct process, would be highly appreciated.

mintar, 2025-01-21:
We are publishing the ground truth pose of the robot as `nav_msgs/Odometry` messages on the `base_pose_ground_truth` topic:

https://github.com/DFKI-NI/mir_robot/blob/28b16c8946d0cc8f22bc429361f1b8b78f74d1cc/mir_description/urdf/include/mir.gazebo.xacro#L48-L61

So, if you replace `odom` with `base_pose_ground_truth`, I hope it does what you want.

huyhoang251296, 2025-01-22:
Thank you. `base_pose_ground_truth` works!!

closed 2025-01-22 by huyhoang251296