DFKI-NI/mir_robot #141: Build URDF-Model from XML Macros

BlackRose01, 2025-01-22:
Hello,

I would like to use that project with Drake. But it requires an URDF file instead of an XACRO file. I tried to build the file with the support of the following page [https://www.kwesirutledge.info/thoughts/ur_urdfs](https://www.kwesirutledge.info/thoughts/ur_urdfs) but it always throws different errors that the `mir_description` package could not be found.

I used the same commands like on the linked page. I also deleted all unnecessary packages from the project and replaced the CMakeLists.txt from this project with the mentioned project on the page.

I hope you can help me.

Thanks in advance.

KR,
BlackRose01

mintar, 2025-01-22:
Install Ubuntu 20.04 and ROS Noetic and follow the README instructions for installation. No need to mess around with the CMakeLists.txt or anything. Then run the following command:

```bash
xacro $(rospack find mir_description)/urdf/mir.urdf.xacro mir_type:=mir_100 tf_prefix:="" > mir.urdf
```

This will generate the file `mir.urdf` with the following content:

```xml
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from /home/mguenther/ros/noetic/mobipick/src/mir_robot/mir_description/urdf/mir.urdf.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="mir">
  <!-- necessary because args cannot be accessed inside ${} expressions -->
  <link name="base_footprint"/>
  <joint name="base_joint" type="fixed">
    <parent link="base_footprint"/>
    <child link="base_link"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
  <link name="base_link">
    <inertial>
      <mass value="77.0"/>
      <origin rpy="0 0 0" xyz="-0.012354000000000004 0.0 0.15"/>
      <inertia ixx="0.8341663330000001" ixy="0.0" ixz="0.0" iyy="1.8608325890000004" iyz="0.0" izz="2.1816657939999997"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0.037646 0 0"/>
      <geometry>
        <mesh filename="package://mir_description/meshes/visual/mir_100_base.stl"/>
      </geometry>
      <material name="white">
        <color rgba="1 1 1 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0.037646 0 0"/>
      <geometry>
        <mesh filename="package://mir_description/meshes/collision/mir_100_base.stl"/>
      </geometry>
    </collision>
  </link>
  <gazebo reference="base_link">
    <material>Gazebo/White</material>
  </gazebo>
  <!-- IMU -->
  <joint name="base_link_to_imu_joint" type="fixed">
    <parent link="base_link"/>
    <child link="imu_link"/>
    <origin rpy="0 0 0" xyz="0.0 0.0 0.25"/>
    <!-- same as real MiR -->
  </joint>
  <link name="imu_link"/>
  <gazebo>
    <plugin filename="libhector_gazebo_ros_imu.so" name="imu_plugin">
      <updateRate>50.0</updateRate>
      <bodyName>imu_link</bodyName>
      <frameId>imu_frame</frameId>
      <!-- from real MiR -->
      <topicName>imu_data</topicName>
      <accelDrift>0.0 0.0 0.0</accelDrift>
      <accelGaussianNoise>0.007071067811865475 0.01 0.011401754250991379</accelGaussianNoise>
      <!-- real MiR linear_acceleration_covariance: [5e-05, 0.0, 0.0, 0.0, 0.0001, 0.0, 0.0, 0.0, 0.00013] -->
      <rateDrift>0.0 0.0 0.0</rateDrift>
      <rateGaussianNoise>0.00282842712474619 0.00282842712474619 0.0005477225575051661</rateGaussianNoise>
      <!-- real MiR angular_velocity_covariance: [8e-06, 0.0, 0.0, 0.0, 8e-06, 0.0, 0.0, 0.0, 3e-07] -->
      <yawDrift>0.0</yawDrift>
      <yawGaussianNoise>0.31622776601683794</yawGaussianNoise>
[... 559 lines cut ...]
      <alwaysOn>true</alwaysOn>
      <updateRate>50.0</updateRate>
      <bodyName>base_footprint</bodyName>
      <topicName>base_pose_ground_truth</topicName>
      <gaussianNoise>0.01</gaussianNoise>
      <frameName>map</frameName>
      <xyzOffsets>0 0 0</xyzOffsets>
      <rpyOffsets>0 0 0</rpyOffsets>
    </plugin>
  </gazebo>
  <gazebo>
    <plugin filename="libgazebo_ros_control.so" name="gazebo_ros_control">
      <controlPeriod>0.001</controlPeriod>
      <legacyModeNS>false</legacyModeNS>
    </plugin>
  </gazebo>
  <!-- instead of the controller_plugin_gazebo, you can use the diffdrive controller like this:
    <xacro:diff_controller_plugin_gazebo prefix=""
      left_wheel_joint="left_wheel_joint"
      right_wheel_joint="right_wheel_joint"
      wheel_separation="${2*mir_act_wheel_dy}"
      wheel_radius="${mir_act_wheel_radius}"/>
  -->
</robot>
```

BlackRose01, 2025-01-27:
Thanks, it worked. For completeness, I used Docker/Podman to build the file. The used image was `osrf/ros:noetic-desktop-focal` and I did everything in the README.md under `Source install` except the last 2 steps. Here then I used your command.

KR,
BlackRose01

closed 2025-01-27 by BlackRose01