DFKI-NI/mir_robot #142: Status of ROS2 support

Cr34t1ngH0p3, 2025-01-23:
Hi, 
I just started with the project and ros. 
I was able to set it up on Ubuntu 20.04 and noetic. 
Is it also possible to set it up on an Ubuntu 22.04 (with humble)?
I saw that you guys created branches and issues for that, but from my view it is not finished. 

Am I right with that?

Thank you for the help

mintar, 2025-01-23:
Unfortunately you're right, it's not finished. I've summarized the current state of the ROS2 support here: https://github.com/relffok/mir_robot/issues/32#issuecomment-1847471919

However, I have plans to finally take some time to work on the ROS2 support soon.

closed 2025-01-24 by Cr34t1ngH0p3

shrutic12, 2025-04-16:
Hello @mintar I was wondering if you have any updates on ROS2 support? Is the code https://github.com/mintar/mir_robot/tree/ros2 ready for use at the moment? Thanks!