DFKI-NI/mir_robot #143: MiR250 control through /cmd_vel

nevesapl123, 2025-04-04:
I'm currently working with a MiR250 robot using ROS and I'm trying to implement continuous velocity control via the /cmd_vel topic.

Right now, whenever I send commands to /cmd_vel, the robot enters safe mode unless I manually interact with the touchscreen interface – for example, by enabling "Remote Control" mode via the joystick UI and pressing the flashing blue button on the yellow remote command

My question is:

Is there any way to keep the robot active and accepting /cmd_vel commands without having to manually enable remote mode on the interface every time or keep touching the joystick from time to time preventing it from going to protect mode?

   - I'm trying to control the robot purely via ROS topics.

   - Ideally, I want to avoid needing to touch the robot or its interface to allow movement.

Is there any setting, parameter, or workaround that can help with this?

Thanks in advance!

mintar, 2025-04-04:
I don't recall having this problem, but I haven't tested it recently. I mostly use a MiR100 (which is almost identical to the MiR250 from a software perspective) with one of the software versions in the README. I mostly interact with it by sending move_base commands, and it has been a while using the cmd_vel interface, so I have to test again, but I guess this is a relatively recent change. Which SW version are you running?

nevesapl123, 2025-04-04:
I'm almost certain that the installed software version is 2.14, although I can only confirm that on monday when I have access to the robot again.

jamd315, 2025-04-04:
I've used both the MiR 100 and MiR 250 and ran into this problem a while ago.  As I remember it, the MiR 100 requires no intervention to enter manual mode, but the MiR 250 requires a safety acknowledgement, and will time it out with no input.  From what I remember of the solution, I think it was to send an HTTP request to the web UI's /login and get a user token, then use a variant of the /cmd_vel topic that accepted that token.  This let me control the robot programmatically, but still required a safety acknowledgement.

nevesapl123, 2025-04-05:
Thanks for the clarification! Makes sense now. The MiR100 is like the cool uncle who lets you drive without asking too many questions, while the MiR250 is more like the reverse 😅

Good to know I can at least automate the overprotective behavior through the REST API. I’ll give that a try next week, really appreciate the tip!

closed 2025-07-25 by nevesapl123